sanjay900/SantrollerConfigurator #71: [Issue]: Can't get RF working with Pro Micro

RE-Gaming, 2023-02-07:
### Please describe your issue.

I switched over to the pro micro to get an RF setup working with the Wii Guitar. I have a transmitter and receiver wired up, doubled checked the pinout to make sure it was correct. And I followed the steps to on the RF tutorial. 

After going through the steps, I have a transmitter that seems to be taking in the Wii Guitar and sending out commands over the SPI lines correctly, but the Rx is just not receiving anything.

Some observations:
- The Rx side's Iref pin is at 1.2V but the Tx side's Iref pin is at 0V
- The CE pin on the RX pin is High at all times, which looks to be correct when in standby mode CE=1, assuming Prim_RX is being set high
- From power on of the RX device, the SPI lines do have some communications occurring in the first 1ms 

### Version

yes

### What operating system are you using?

Windows 10

### What microcontroller are you using?

arduino pro micro 3.3V

### How is your controller programmed?

Wii

sanjay900, 2023-02-07:
RF is just a pain at the moment. its been a while since i've had that working on this firmware, but something I intend to do is improve the RF stuff in the new firmware once i get around to implementing it again.

RE-Gaming, 2023-02-07:
Is there a prior version I should use instead?

sanjay900, 2023-02-08:
Like it should work as it is, it's just for some people it works and for
some it doesn't not really sure why

How are you powering both units?

RE-Gaming, 2023-02-08:
Once they are programmed, the Rx is powered by my computer and the Tx is plugged into a usb wall charger. The Onboard 3.3V LDO of the pro micro is bring used for the nRF24L01 and the Wii I2C communication.

sanjay900, 2023-02-08:
Odd, i dont see why that doesnt work then tbh, does the config tool still let you configure the receiver once its programmed as a receiver?

sanjay900, 2023-02-09:
you didnt get this error did you?
[image]

sanjay900, 2023-02-09:
assuming your using the latest firmware, i'm surprised some people are having issues and some arent

RE-Gaming, 2023-02-10:
Okay, so I redid the programming process, trying to be as detailed as I can:
1. Plugged in the Rx, kept Tx unplugged
2. Programmed the Rx, then clicked "Start Configuration"
3. immediately went to "configure RF", enabled it and clicked "Program RF Transmitter"
4. Leaving the Rx plugged in, I now plug in the Tx
5. I select it from the drop down and it immediately says "Start configuring" rather than "Start programming". The dialog and 100% status is from the Rx board, I've noticed the text is retained between programming
[image]

6. I select "Start configuring" since its my only option
7. I am presented a blank slate and it still shows the Rx device (com6) rather than the Tx device I selected (com4) in the upper left corner
[image]

8. I still continue with the configuration: Wii input, Ps3 RB Guitar output, pin 9 as digital tilt. But the digital tilt selection doesn't work... and I can't write anything the program freezes.

This is not the behavior I was having a few days ago... So I completely wiped the Guitar Configurator software from my computer and reinstalled.
1. With a fresh install when I plug in my Rx device I see
[image]
2. I select this board, and restore it back to Arduino so I can get a clean slate, now I see COM6
[image]
3. I select COM6 (the other board is currently unplugged), I program it (I see the flashing LEDs on board) then I select "Start Configuration"
4. It must remember my prior settings on this device as it still shows the previous configuration I had even after restoring and reprogramming it?
[image]
5. I go to "Program RF", enable RF is already selected, and I click "program transmitter"
6. I plug in the Tx board (COM4) and I see this screen. Again i'm not presented with a program option, just "start configuration"
[image]
7. I click "Start Configuration" and my settings from the Rx board are already there, its already configured how it want it to be except for tilt which isn't showing the Micro board so I can select a pin
[image]
8. so I click write. And it appears to write to the board, I don't see any LEDs flashing on either board but the program resumes after a brief hang up and the "you have unwritten changes" message is gone

sanjay900, 2023-02-10:
the fact that its showing "Wii Unknown Extension" there is interesting, that would indicate the transmitter cant talk to your wii extension

RE-Gaming, 2023-02-10:
That is interesting, the Pro Miro is sending data over SPI to the nRF when a button is pressed, so it is at least trying. A little hard to tell if it is the correct data. What should it be showing other than Wii Uknnown Extension?

RE-Gaming, 2023-02-10:
I'm going to do an EEPROM Clear and setup some example code for the Pro Micro + nRF in Arduino to confirm its not any of the boards. I'll follow up once I'm done with that and see if there is any different behavior.

Quick questions; does the nRF have to be connected during the configuration setup or can it be soldered on later? I have other Pro Micros available to do testing with but I'd rather not resolder everything if I can avoid it.

sanjay900, 2023-02-10:
> That is interesting, the Pro Miro is sending data over SPI to the nRF when a button is pressed, so it is at least trying. A little hard to tell if it is the correct data. What should it be showing other than Wii Uknnown Extension?

assuming you have a guitar plugged in, wii guitar or something to that effect

sanjay900, 2023-02-10:
> I'm going to do an EEPROM Clear and setup some example code for the Pro Micro + nRF in Arduino to confirm its not any of the boards. I'll follow up once I'm done with that and see if there is any [...]

so the way the transmitter is configured, is it is done via the receiver. Once you have programmed the transmitter, you don't need them both plugged in, and in fact i have in the past had weird issues doing that, so might be worth trying to run the transmitter on batteries or another power supply and seeing what that does.

RE-Gaming, 2023-02-10:
> assuming you have a guitar plugged in, wii guitar or something to that effect

Just to clarify the Tx always show up as COM4, its the Rx that is showing up as Wii Unknown Extension, i'm not sure if that makes a difference.

sanjay900, 2023-02-10:
The stuff showing up on the RX is from the TX side.
In fact, the fact that your seeing anything at all means they are both talking to eachother in theory

RE-Gaming, 2023-02-10:
That makes sense. Okay, let me try to get everything cleared out and double checked then I'll come back with an update.

RE-Gaming, 2023-02-13:
So after running an eeprom_clear and putting together some example code for the nRF24L01, the Tx and Rx devices work in the Arduino environment. I tried programming them again with the Guitar Configurator and I got the same result as above. 
- Listed as Wii Unknown Extension
- The Tx is reacting to button presses correctly and sending SPI commands
- The Rx is running through some setup code correctly
- But there is no response from the Rx when the Tx sends SPI commands

RE-Gaming, 2023-02-13:
One thing I noticed is when I got to program the Tx, I never actually go through the "Program" function, it skips right to configuration which I thought was odd.

sanjay900, 2023-04-03:
since update images is where its at at the moment
[image]
[image]

Finally started implementing RF again yesterday.
Two major improvements are that you can now have multiple rf transmitters connected to one receiver
And the tool will tell you if it can both see that the nrf module is connected correctly, and if it sees the other side of the connection.

sanjay900, 2023-06-13:
not going to lie, the more i think about it the more im very tempted to just drop all this RF support and make the pico W the only way to do wireless stuff

sanjay900, 2023-06-13:
theres just too many bad nrf modules out there and i dont want to be providing support for that

RE-Gaming, 2023-06-29:
Hi Sanjay,

I am so sorry, I haven't checked my github in some time and didn't see
these messages. I will test this out this weekend.

I honestly agree with you, even in the pack of 6 nRF boards I bought 1 was
DOA and 1 has started to act finicky in another project.
I agree with the switch to Pico W, the cost structure of the 1st party
Picos in general is much better than even the Arduino clones...

Nick

sanjay900, 2023-06-29:
Yeah, the more i look into it the more these modules just dont seem worth it. It also means i can just drop support for arduino pro minis as well which simplifies things on my end.

sanjay900, 2023-06-29:
Yeah okay, i will be dropping RF support, its just such a mission to keep that working.

closed 2023-06-29 by sanjay900