sanjay900/SantrollerConfigurator #72: [Request]: Support for RB4 Solo Frets

vcapellan, 2023-02-13:
### Please describe your request.

Hello!

I haven't gotten around to modding my guitar yet (my PS4 RB4 Stratocaster is already disassembled and awaiting my pi pico & soldering kit to come in the mail!). However, going through the guides you have and seeing screenshots of the GUI, it looks like there isn't support for the RB4 solo frets. Are there any plans for supporting this in the future?

Thanks for the hard work! 

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2023-02-13:
already in the new firmware, just need to wait for me to release it

sanjay900, 2023-02-13:
I've implemented it for xbox 360, ps3 and xbox one as well fyi

sanjay900, 2023-03-16:
Ill also likley support using the GH tap bars as upper frets, i have some stuff in the works for making the GHWT tap bar work better

sanjay900, 2023-03-28:
Have just done even more work towards solo frets, specifically so that they are fully supported when using bluetooth or RF modes

sanjay900, 2023-03-29:
If you just want to wire the guitar up for the future firmware, wire the solo frets up to seperate pins and you should be good for the new firmware when it comes out.