sanjay900/SantrollerConfigurator #74: [Request]: Add Support For External Overdrive/Star Power Switch

bmasephol, 2023-02-20:
### Please describe your request.

In the new firmware rewrite you are working on it would be great if you could add support for an external Overdrive/Star Power activation switch.

I've been testing the heck out of my Fender Wireless Bass rebuild I did and on a Raspberry Pi Pico this is solid and works great.  I can't wait for your rewrite release.  If you need testers for that let me know.

I'm waiting patiently for the reworked whammy calibration as well :)

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2023-02-20:
Use the current tilt stuff and wire your button in as digital tilt?

closed 2023-02-20 by sanjay900

bmasephol, 2023-02-22:
Yes, not sure why I didn't think of that first time around.  Keep up the good work!