sanjay900/SantrollerConfigurator #76: [Issue]: "GHTV" Button (LS Click) does not show up in configurator

ujicos, 2023-03-04:
### Please describe your issue.

This renders the entire GHTV mode unplayable. I couldnt figure out how to build this as i am on windows, but this should be a simple fix as looking at the code the support is already there. Hovering on the button in the graphical preview does NOTHING, nor is it in the list view

### Version

yes

### What operating system are you using?

Windows 11

### What microcontroller are you using?

arduino pro micro 5V

### How is your controller programmed?

Direct

sanjay900, 2023-03-04:
Guitar hero live support is broken in console anyways iirc

ujicos, 2023-03-04:
> 

GHTV Reloaded is a community project that revived GHTV for RPCS3, Real PS3 Hardware, and Real Wii U Hardware.
Could you push a update so i can enable this button? It's really essential for me, and as far as i can tell, the support is already in the source code

ujicos, 2023-03-04:
Alternatively, provide a way to generate a solution file so i can fix this myself

sanjay900, 2023-03-04:
yeah no the source code you see there is completly irrelevant to the current build of the configuration tool. The source code you can see is unfinished and unusable.

ujicos, 2023-03-05:
> 

I had tilt working just fine. All i need is the GHTV button to work.

Isn't this just the current 9.1.2 build source? a.k.a the same as whats up right now?

sanjay900, 2023-03-05:
not even close

ujicos, 2023-03-05:
Alright, Thank you very much.
I didnt intend to come off as rude in the case i did.

sanjay900, 2023-03-05:
all good, you aren't the first one to ask for this, so i really should have done it earlier. Was hoping this firmware rewrite would be done by now but its taking longer than id like

sanjay900, 2023-03-05:
If your curious, this is what the 9.1.3 is based on: https://github.com/sanjay900/guitar-configurator/tree/qt

ujicos, 2023-03-05:
> If your curious, this is what the 9.1.3 is based on: https://github.com/sanjay900/guitar-configurator/tree/qt

I'll have a look. For now i am waiting for the build to succeed for Windows, since github decided it didnt want to build it lol

sanjay900, 2023-03-05:
You can probably see why im trying to migrate away from this firmware lol

ujicos, 2023-03-05:
> You can probably see why im trying to migrate away from this firmware lol

I would say yes, but i have zero experience in Qt.
I couldnt even start compiling after grabbing source, lol

sanjay900, 2023-03-05:
this pretty much happens every time QT updates, and since its been a year since the last update there have been a few updates. For some reason its being more difficult than normal though

sanjay900, 2023-03-05:
https://github.com/sanjay900/guitar-configurator/releases/tag/v9.1.5
that took more effort than i would have liked haha

closed 2023-03-05 by sanjay900

ujicos, 2023-03-05:
I regret to inform you the GHTV button is still not being recognized. Maybe i was wrong about it being LS?
The graphical view also doesnt have the button showing up, only list view, but i really dont care for that, just more-so mentioning it so you're aware.