sanjay900/SantrollerConfigurator #77: [Request]: RP2040 Zero compatible?

Namrepus221, 2023-03-21:
### Please describe your request.

Is the RP2040 Zero board compatible to use as a board for the guitar configurator or is it underpowered?

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

N/A

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2023-03-21:
Try it

sanjay900, 2023-03-25:
Funny story, a mate of mine just asked me if i wanted an RP2040 zero, so i guess when that ships i will have one to test.

Namrepus221, 2023-03-25:
Only reason I’m asking is it only has 2 GND pins and was worrried it would cause problems with wiring

sanjay900, 2023-03-25:
you can wire as many things to a single ground as you want.

closed 2023-03-25 by sanjay900