sanjay900/SantrollerConfigurator #79: [Request]: Authentication on Xbox 360 without RGH

CristobalPenaloza, 2023-04-11:
### Please describe your request.

Hello! This project is amazing, and I want to direct mod my wii guitars and drums (no plan on going back to wii).

I want to use them on Xbox 360.
But my 360 is a winchester and can't be modded, and I currently have no way of getting another one that is compatible with RGH.

So I'm stuck on the console's controller authentication.
I know that is possible to bypass that authentication using a middleman, like a cronusmax. But that's **extremely** expensive.

Could there be a way to use the very same pico of the guitar to skip the authorization?
Something like, including an extra usb port on the microcontroller to connect an og controller for the authorization. Is a setup like that possible with this firmware?

Thank you so much in advance!

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2023-04-12:
Its already implemented in the firmware in development, im just still working on finishing it. Its implemented for Xbox 360, Xbox one and PS4.

sanjay900, 2023-04-12:
There's support in the firmware for automatically working out what console a device is plugged into too, so no more programming for different consoles. I also have some code in there for forwarding inputs from a plugged in controller too, allowing for making your own xim or cronusmax.

CristobalPenaloza, 2023-04-12:
That sounds amazing!
There is currently no plan on releasing an experimental build for the new firmware?
Like, to be able to test those new features, even if it's not fully finished?

sanjay900, 2023-04-12:
I need to get the new tool working better before I can do that.

sanjay900, 2023-04-13:
The other problem is so much is changing with the firmware currently that having an older firmware installed can be unstable with a newer tool, which isnt something i want to deal with until everything is going.