sanjay900/SantrollerConfigurator #82: [Issue]: Keebio Elite-Pi does not recover after unplugging and replugging

dakotaaesoph, 2023-04-23:
### Please describe your issue.

After configuring, physically unplugging the device from the computer and replugging it in, the microcontroller does not show back up in Windows. I have to use the reset button on the microcontroller board and reconfigure the guitar.

Everything seems to work fine until I unplug and replug the USB in. Looking at a USB event viewer, it doesn't even show up in Windows at all. 

This is the board I used: https://keeb.io/products/elite-pi-usb-c-pro-micro-replacement-rp2040

### Version

yes

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2023-04-23:
Means you need to pick one of the other rp2040 based boards, might i recommend the KB2040 first as that one seems to be the most compatible

dakotaaesoph, 2023-04-24:
Perfect, that fixed the issue. Thank you!

closed 2023-04-24 by sanjay900