sanjay900/SantrollerConfigurator #83: [Issue]: can't enter configuration

Fifroo, 2023-04-25:
### Please describe your issue.

i entered the configuration only one time every next time its stuck on flashing, when its finished its automaticly detected as leonardo/pro micro xinput and i cant configure it
[image]

### Version

yes

### What operating system are you using?

Windows 11

### What microcontroller are you using?

arduino pro micro 5V

### How is your controller programmed?

Direct

Fifroo, 2023-04-25:
on second pc its working idk what is wrong on mine

sanjay900, 2023-04-25:
So this is a weird windows thing that i can't easily fix, but it should be fixed in the new tool im working on. I can't really trace what causes it though.