sanjay900/SantrollerConfigurator #84: [Request]: Is this possible to make a drum with a PI PICO?

Namoku, 2023-05-06:
### Please describe your request.

Hello, I tried to make a drum controller with an old RockBand wireless drum I had, but after tried to flash the arduino uno, it doesn't pass 'update needed' in the guitar configurator, so Imma try to make it with a PI PICO, but here's my question, how can I connect the piezoelectric sensors to the board? In the arduino I connected in the analog input using 1M resistor.

### What operating system are you using?

Windows 11

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2023-05-06:
Not yet no since the pico doesnt have enough analog inputs, some work is being done for that.
Try some of the older firmwares with your uno and see if any of them work.

sanjay900, 2023-05-07:
https://github.com/sanjay900/guitar-configurator/releases/tag/v9.1.16
This should fix your uno

Namoku, 2023-05-19:
Thank you, it fixed but my drums is not working so Imma give it a try later

sanjay900, 2023-05-19:
Im not really sure how functional drums were on that firmware in the first place tbh, i never made a set myself. I know some people got it working, and some didnt, so your mileage may vary with getting that working.