sanjay900/SantrollerConfigurator #85: [Issue]: Arduino Uno Out of Date | Program Crashes when "Start Configuring"

TripleTell, 2023-05-07:
### Please describe your issue.

Following this guide: https://sanjay900.github.io/guitar-configurator/guides/wii.html#steps
I am having trouble at step 11. 

Whenever I click "Start Configuring," the program closes/crashes.
[image]

Extra information that might be useful.

- When I start from the beginning, I do not have to put the board back in DFU mode. It automatically recognizes the board in DFU mode and continues unprompted.

- I have tried it with two separate Arduino Uno boards I had laying around, and I have the same outcome.

- I was able to get to the Configuring part the first time, but it did not have a picture of the guitar like the guide had.

- My Device Manager recognizes the Arduino as Ardwiino

- [image]

### Version

yes

### What operating system are you using?

Windows 10

### What microcontroller are you using?

arduino uno

### How is your controller programmed?

other issue not controller related or I have not gotten this far

sanjay900, 2023-05-07:
Yeah i need to rebuild the uno code i think

TripleTell, 2023-05-07:
Which older code should I use?

sanjay900, 2023-05-07:
Anything before the xb360 fixes

sanjay900, 2023-05-07:
Trying to release a fixed version but for some reason the macos builds have just stopped working out of nowhere :(

sanjay900, 2023-05-07:
https://github.com/sanjay900/guitar-configurator/releases/tag/v9.1.16

closed 2023-05-07 by sanjay900

TripleTell, 2023-05-07:
Thanks so far! I downloaded v9.1.16 and the next issue arrises when i click "Write".

[image]

My computer then has the sound alert as something disconnected and displays a "Controller disconnect" on the screen.

[image]

Thoughts?

sanjay900, 2023-05-08:
does it not connect back?

sanjay900, 2023-05-08:
tbh, it looks like its all programmed anyways, what setting are you trying to write at the moment?

TripleTell, 2023-05-08:
I don't know if it connects back or not. Should the program do anything different? I went into Clone Hero and the guitar wasn't recognized or working.

sanjay900, 2023-05-08:
if you look in the top left corner of the tool, it says "wii no extension"
this means your wii guitar isnt being detected and you have something wired up wrong.

sanjay900, 2023-05-08:
oh, which vcc did you use on the uno?

sanjay900, 2023-05-08:
Which breakout are you using?

TripleTell, 2023-05-08:
No breakout. Using wires to go directly from the Wii plug to the Arduino. Low quality photo below. 

[image]

sanjay900, 2023-05-08:
I updated the instructions, your wiring isnt quite right.

sanjay900, 2023-05-08:
If this continues to not work, then the problem may just be you poking wires at it like that, i know a lot of people that couldn't get that method working because the wires just didnt make good contact.

sanjay900, 2023-05-08:
Actually, can i see a picture of the connector in the guitar neck please?

TripleTell, 2023-05-08:
[image]

sanjay900, 2023-05-08:
Have you got discord by any chance? If so add me, sanjay900#0001

TripleTell, 2023-05-08:
The neck is plugged in when I upload and test. I added on discord.