sanjay900/SantrollerConfigurator #88: [Request]: Make it possible to change pin binding of mosl and sck for leds

RatInDaHat, 2023-05-31:
### Please describe your request.

I'm hoping it would be possible to change the pin binding for sck and mosl for leds. I wired my current board differently than recommended with sck on GPIO 18(pin 24) and mosl on GPIO 19(pin 25). I'm using a rp pi pico. I didn't see where the pin binding is happeneing in your code so here i am asking.  

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2023-05-31:
it will be possible in a future version of the tool

sanjay900, 2023-05-31:
I had to rewrite everything from scratch to add this sort of thing - the old firmware was really designed around the arduinos that didnt have configurable pins for that sort of thing, so ive been working on a new one for the past year or so that can let you bind absolutly everything on the picos.

sanjay900, 2023-05-31:
[image]

RatInDaHat, 2023-05-31:
That is really awesome, and I'm sure a lot of work. When will it be ready for prime time?

sanjay900, 2023-05-31:
ive got no idea

RatInDaHat, 2023-05-31:
I'll keep my eye open for it in the future. Thanks