sanjay900/SantrollerConfigurator #89: Problem creating RF guitar :(

MeinzRamone, 2023-06-09:
### Please describe your issue.

I have a problem when I want to record the RF module, the application simply gets stuck and closes by itself.....when I flash the rf module, 3 panels open saying that arduinos were detected...I thought that the arduino was the bad one but I tried the 6 new ones and the same thing... I tried another pc and the same thing... I had the illusion of creating a wireless guitar :(

### Version

yes

### What operating system are you using?

Windows 11

### os-other

Windows 10

### What microcontroller are you using?

arduino leonardo 5V

### How is your controller programmed?

other issue not controller related or I have not gotten this far

sanjay900, 2023-06-09:
yeah RF is a bit busted at the moment. Im working on something new there but its not ready yet.
Tbh, imo id probably just recommend a pico w + bluetooth for when the new version comes out, its just far easier than rf

sanjay900, 2023-06-09:
hit me up on discord if you want, same username, and i might be able to get you some testing firmware

sanjay900, 2023-06-29:
Yeah okay, i will be dropping RF support, its just such a mission to keep that working.

closed 2023-06-29 by sanjay900