sanjay900/SantrollerConfigurator #90: [Request]: Support for ItsyBitsy 32u4

okennedy, 2023-06-27:
### Please describe your request.

Although there appear to be references to the itsybitsy in the guitar configurator source code, the device itself does not appear to be getting detected properly.  The 32u4 variant [appears to be](https://www.adafruit.com/product/3677) compatible with the Leonardo, so unless I'm mistaken, the only change needed is [detecting the device's Vendor/Product ID](https://github.com/sanjay900/guitar-configurator/blob/master/Board.cs#L123).  

For the IDs, lsusb gives:
```
Bus 001 Device 014: ID 239a:800e Adafruit ItsyBitsy 32u4 5V 16MHz
```

I'll try to wrangle together a PR when I get a chance.

### What operating system are you using?

Linux

### What microcontroller are you making a request for?

arduino leonardo 5V

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2023-06-27:
yeah, any PR you do here wont help you as this is a totally different codebase, but doesn't the tool let you program generic serial devices anyways?

sanjay900, 2023-06-27:
All you might have to do is throw it into bootloader mode first, but thats not too hard

sanjay900, 2023-06-27:
You can also just program it with the arduino IDE first, as a pro micro and that will let the tool pick it up

sanjay900, 2023-06-27:
interesting, it used to support any serial device i wonder when that broke

sanjay900, 2023-06-27:
Itll be supported once v9.4.5 comes out.

closed 2023-06-27 by sanjay900