sanjay900/SantrollerConfigurator #92: [Issue]: Whammy calibration not working with Joystick - Deadzone issue?

someot, 2023-07-04:
### Please describe your issue.

Attempting to use a Joystick as a whammy bar seems to be busted due to the calibration process.

If I used a Joystick Y axis and calibrate with the minimum and maximum being the corresponding min/max for the +/- axis, the deadzone looks to assume positive only and starts from 0 up. 
[image]

You need to then move the deadzone all the way to 0, which gets you _close_ but not good enough.
[image]
Joystick at neutral.

If you instead set up the whammy so that minimum value is Joystick at rest and Maximum is full movement to on positive axis (deadzone setting at rest), you will end up with a whammy value always at negative.
[image]

And you'll be forever stuck with a permanently pressed
[image]

### Version

yes

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

someot, 2023-07-04:
I pulled down and built master (like this new version a lot btw), it looks to have the same issue. I'm not an expert on these things but my guess is it has something to do with the whammy config using 0 as the at rest value. 0 is interpreted as full moment on the negative axis on the emulated controller, so even when the physical joystick is at rest, it sends 0. That causes the emulated joystick to read as pushed all the way in.

someot, 2023-07-04:
Maybe that theory doesn't hold any weight though, cuz assigning as joysticks for gamepad config does not display this behavior.

[image]

sanjay900, 2023-07-04:
joysticks and whammy use  different deadzone handling

someot, 2023-07-04:
Again, I'm no expert but I wonder if this can be worked around by outright mapping the right joystick and no whammy. Not sure if that's possible with the way a guitar is bound, but the XInput bindings seems to assign whammy to right joystick anyway.

sanjay900, 2023-07-04:
I mean i can make this possible, i just dont see it working very well as whammy if i do

someot, 2023-07-04:
Yup, mapped as gamepad and it worked perfectly.

Here's how they ended up (config exported from master, pulled today).
[xbox bindings.zip](https://github.com/sanjay900/guitar-configurator/files/11944680/xbox.bindings.zip)

sanjay900, 2023-07-04:
so the thing is, what your saying as "perfectly" isnt what a game is expecting, and thats the problem

someot, 2023-07-04:
Perhaps we've gotten down a rabbit hole and I should restate the problem. 

The issue that occurs with the current state of joystick as whammy with the configurator is that the XInput emulation sends the Right Thumbstick X axis as fully negative for a value of "0" (as reported by the configurator). Doesn't matter what you set up, if the joystick is at a position considered to be 0 by the configurator, XInput reads it as full negative (which is right thumbstick pressed all the way left).

 This means that anything that has a control for the right thumbstick is always thinking it's being pushed fully to the left.

sanjay900, 2023-07-04:
> 

This is what a real guitar does as well.

someot, 2023-07-04:
That's great but this isn't a real guitar

sanjay900, 2023-07-04:
But you are trying to emulate one?

sanjay900, 2023-07-04:
as youve found, the gamepad works because you are no longer trying to emulate a guitar at that point, which makes sense as the problem you have stated is specifically due to how guitars work. This isn't an issue with you using a joystick, this is just how guitar emulation works.

someot, 2023-07-04:
Perhaps... the issue is with how Clone Hero is reading the input?

sanjay900, 2023-07-04:
what exactly is it doing in clone hero?

someot, 2023-07-04:
Any time you hit a long note, it's auto "whammied" to a flat level, which is probably 50% now that I think about it. I'm gonna test something real quick, I think I might have the solution.

sanjay900, 2023-07-04:
What happens if you calibrate it, so that the middle of the joystick is the bottom of the whammy, and then pushing it right is the pushed in state

sanjay900, 2023-07-04:
i am kinda curious though - why are you trying to use a joystick as whammy?

someot, 2023-07-04:
> What happens if you calibrate it, so that the middle of the joystick is the bottom of the whammy, and then pushing it right is the pushed in state the game _should_ be expecting all the way left as no whammy and all the way right as full whammy by default so of course, if you set it in gamepad mode, then your joystick sitting in the middle would result in 50% whammy

I think this is mostly it, except that CH _is_ picking this up as a guitar (when configured that way). But for some reason, the default whammy configuration is not as you stated.

> i am kinda curious though - why are you trying to use a joystick as whammy?
Built the Polybar. Doesn't have anything built in for whammy so I wanted to figure something out because I've always loved the whammy effect.

sanjay900, 2023-07-04:
CH is kinda weird in that once you play with the calibration once, the defaults just no longer work correctly, and im not really sure why it works that way, but it is something i have found as well myself.

someot, 2023-07-04:
Yeah, I'll just have to play around with it to get things feeling right. That's probably it for this issue, thanks for your time. I really like the new version of the configurator so keep up the good work.

closed 2023-07-04 by someot

sanjay900, 2023-07-04:
Should be easy enough to just calibrate it within CH to work correctly

sanjay900, 2023-07-04:
out of curiousity, what does your calibration look like within CH?

someot, 2023-07-04:
Ok after going into this deeper I think I'm going to throttle my partner. The whole reason I went down this was because she kept complaining about the whammy being on. First thing I asked was "well does it happen on your PC _and_ the SteamDeck, or just one? She said both and I trusted her. She was wrong. It only happens on the SteamDeck. Because the SteamDeck picks it up as a generic axis and sets the deadzone like a generic joystick. I'm mad.

sanjay900, 2023-07-04:
Im just glad you figured out what was going on