sanjay900/SantrollerConfigurator #93: [Issue]:  When using drums on PS3 inputs are randomly dropped / not registered

darkziel, 2023-07-07:
### Please describe your issue.

I am using a Wii Band Hero Drum Kit connected to a RP, on the guitar configurator I set the output to PS3 and the game recognizes the drums but when hitting the drums sometimes it randomly does not register the input. 
When using the exact same setup on the PC without changing any configuration no inputs are dropped so I think there must be an issue on the PS3 side, do I need to use a specific configuration on the PS3?
Using Wii guitars works fine and I have not had any issues, only the drums are very unreliable when used on the PS3, I used a gamepad tester homebrew on the PS3 and when hitting the pads it randomly registers the DPad right, and sometimes it doesn't register any input at all.

### Version

yes

### What operating system are you using?

Other

### os-other

PS3 FW 4.90

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Wii

sanjay900, 2023-07-07:
Which game are you testing with?

JR2703, 2023-07-07:
Glad I'm not the only one I'm using the exact same drum kit. Little to no input response on the pads as well. Works perfect on clone hero

sanjay900, 2023-07-07:
Can i get either of you to hit me up on discord? Ill have to get some things tested by ya. Same username

Cicabe, 2023-07-10:
Also an issue I was experiencing. Unfortunately can’t be part of any testing yet since GC isn’t recognizing my drums at all anymore (though I suspect this is more an issue with my drum brain than with the software). But if it’s at all possible to get the drums to be read correctly by PS3, I’d be greatly interested once I get my brain replaced.

sanjay900, 2023-07-11:
If your going to be replacing your brain anyways, wonder if you would be better off just testing hardwiring instead of buying a new brain

Cicabe, 2023-07-11:
The brains just plug right in though, at least on the Band Hero models. Plus, I wouldn’t know how to hardwire them.