sanjay900/SantrollerConfigurator #94: inverted inputs

memes4daysz9, 2023-07-07:
### Please describe your issue.

before i started to solder anything in, i checked if it worked for PS3, when i did so (i was already in a song before i connected) the game would resume with all the frets being pressed while the board isnt connected to anything, then i tried on clone hero, same issue ocured. there is no bridging connections, i couldnt find anything on how to fix this in the issues area, i apologize if this is a clone of someone elses problem

### Version

yes

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2023-07-07:
weird, never had that happen to anyone before. Are you saying nothing is connected at all and this is happening? You haven't turned on LEDs have you?

memes4daysz9, 2023-07-08:
yes, when nothing is connected, all the buttons that have been configured are pressed, and the LED's have not been configured as im not going to be using them

sanjay900, 2023-07-08:
i just can't think of how this would happen, since your the only person this has happened to. What happens if you short the pin you configured to ground or to vcc?

sanjay900, 2023-07-08:
If you have discord feel free to contact me on there and then i can figure this out quicker, since its much easier to debug things that way

memes4daysz9, 2023-07-08:
the problem was a broken resistor rendering the device needing to use the prerelease version (currently 9.4.7 at given post)  

how to fix this if you have this same issue:
 for now, this is the prerelease version: https://github.com/sanjay900/guitar-configurator/releases/tag/preview

click the clear all button when configured inside the config screen, then click the "add binding" button, put it to any button

press the "find pin" in the "Pin" row, you may need to press this twice if it automatically goes to another pin 

then short the pin you want to use but instead of using ground you want to use the 3V3 Pin 

after you do that press the drop down named "Mode" to Advanced instead of standard mode

then that should be it!

closed 2023-07-08 by memes4daysz9