sanjay900/SantrollerConfigurator #97: [Issue]: Install on Windows 10 takes an extremely long amount of time and is riddled with errors

Soundtoxin, 2023-07-23:
### Please describe your issue.

I bought a pre-programmed adapter and had yet to use the configurator. I had the seller set it up for PS3, but I recently RGH-modded an Xbox 360 and wanted to test the guitar on there (I already got usbdsecpatch working and verified my Flatbox works on the 360 now), which will need XInput instead of DInput I believe.
I went to install the latest stable release on a Windows 10 PC and keep getting similar errors about either a file path not existing or not being allowed to write somewhere. The disk is not full. I did not change the default install path from C:\Program Files\whatever.
I noticed what appeared to be some slashes in the wrong direction in the middle of the path. I found a similar complaint in #34 but it looks like you said you hadn't seen it before and so it was maybe not fixed properly. I got too many errors to keep taking screenshots of them and I don't want to upload all of the ~13 or so I took here since that seems spammy, but here are a few of them. I didn't time exactly how long I spent waiting on the installer but it felt unusually long. This machine has an SSD and an adequate amount of RAM.

[image]
[image]
[image]
[image]

I understand you're working on replacements for the configurator and the firmware, but they don't appear to be ready yet. The new configurator at least had a binary I could download, but I suspect it's meant for use only with the new firmware, which I didn't see any prebuilt images of that I could flash to a pico. I haven't tried an older version yet to compare, and I haven't tried installing on another computer/OS yet, but may end up trying that if necessary. This computer is not my main one, it's just in a convenient spot for testing things.

### Version

yes

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2023-07-23:
Are you possibly running out of space or something? Otherwise maybe it just didnt download correctly, i can't really see what else would cause this

sanjay900, 2023-07-23:
I don't actually think the path names are an issue, windows sorta just handles that usually, it almost just seems like its not able to create some files for some reason, and i wouldn't know the cause of that

Soundtoxin, 2023-07-24:
After reinstalling with a non-default path (straight to `C:\`) the program was able to work. Also ran as administrator this time, unsure if that mattered or not. I'll close this issue for now.

closed 2023-07-24 by Soundtoxin