sanjay900/SantrollerConfigurator #98: [Request]: Allow changing of pins for SCL and SDA for Wii Adapter

Ham5andw1ch, 2023-07-25:
### Please describe your request.

I have a RP2040 compatible board that is missing pins 17 and 18. Can we have a configuration option to change which pins map to SCL and SDA for making a wii adapter.

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Wii

sanjay900, 2023-07-25:
already implemented in the new firmware

sanjay900, 2023-07-25:
[image]

jamesfoley, 2023-07-28:
Just about to request the same thing, cool to see it has already been added. Is the new firmware available in any pre-release state, or do you have a rough release date for it?

sanjay900, 2023-07-28:
No pre release yet, no release dates either