sanjay900/SantrollerConfigurator #99: [Issue]:

bartek009w, 2023-07-28:
### Please describe your issue.

Hello,
I have a problem with setting up controller for guitar hero on wii. Will it work with GH? Only choice in programming is rock band.

### Version

yes

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Wii

sanjay900, 2023-07-28:
GH doesn't use USB guitars

sanjay900, 2023-07-28:
This is a thing for modded consoles but it's very in development
https://github.com/sanjay900/fakemote

bartek009w, 2023-07-28:
I plug the guitar into the wii remote controller, GH detects this by changing the tag under options from A and B in circle to color of frets from guitar, but it doesn't work at all.

sanjay900, 2023-07-28:
What exactly are you trying to do, are you trying to use a modded guitar on a Wii to play GH, or use a Wii GH guitar on a PC with this software

bartek009w, 2023-07-29:
Im trying to use modded Guitar to play GH on Wii.

sanjay900, 2023-07-29:
Yeah thats not a thing you can do. Not how it works, the GH games don't support USB. You can try fakemote if you want, which will let you emulate a wii remote with a usb guitar, but its still in testing.

closed 2023-07-29 by sanjay900